remne/p1ib #4: MQTT reconnection behavior

NateRobinsonS, 2022-03-07:
Installation and setup worked as expected.  However, I've noticed that restarting the MQTT broker (Eclipse Mosquitto) causes the connection with the P1ib to break.  It appears that the P1ib doesn't reconnect to the broker automatically, but needs to be restarted via its web interface.  Upon restarting, the P1ib resumes publishing notices to the broker as expected.  More precisely, Home Assistant doesn't find new data from the P1ib.  (I haven't used another method to monitor the MQTT broker directly.)

Currently running the development firmware (f7b009f) on the P1ib.

remne, 2022-03-07:
Hi!
I did a test from my side, and it works as expected with reconnection to the MQTT  broker.

Output from usb/serial port, "<---- <test>" are my added comments:

uptime:406716 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:411717 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:416718 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:421719 ms         IP 192.168.0.217        WiFi TX: 72/78
MQTT: Attempting connection without user/pass...                       <--- MQTT broker shut down by me
[E][WiFiClient.cpp:258] connect(): socket error on fd 59, errno: 104, "Connection reset by peer"
MQTT: failed, Connection Failed
uptime:426720 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:431721 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:436722 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:441723 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:446724 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:451725 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:456726 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:461727 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:466728 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:471729 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:476730 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:481731 ms         IP 192.168.0.217        WiFi TX: 72/78
MQTT: Attempting connection without user/pass...                      <-- MQTT reconnect attempt after 1 minute, broker still shut 
                                                                                                           down by me.
[E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 104, "Connection reset by peer"
MQTT: failed, Connection Failed
uptime:486732 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:491733 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:496734 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:501735 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:506736 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:511737 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:516738 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:521739 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:526740 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:531741 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:536742 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:541743 ms         IP 192.168.0.217        WiFi TX: 72/78
MQTT: Attempting connection without user/pass...                 <-- MQTT broker started, and connection was successful during 
                                                                                                      next connection attempt.
MQTT: connected!
uptime:546744 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:551745 ms         IP 192.168.0.217        WiFi TX: 72/78
uptime:556746 ms         IP 192.168.0.217        WiFi TX: 72/78
Resetting buffer                                                                           <-- P1 telegram received
Start found!
Stop found!
Received bytes: 581
Complete packet received!
{
  "0.0.1.0.0.255": "2022-03-07T21:35:50+0100",
  "1.0.1.7.0.255": 3.891,
  "1.0.2.7.0.255": 0,
  "1.0.3.7.0.255": 1.283,
  "1.0.4.7.0.255": 0,
  "1.0.31.7.0.255": 7.5,
  "1.0.51.7.0.255": 5.3,
  "1.0.71.7.0.255": 4.6,
  "1.0.32.7.0.255": 233,
  "1.0.52.7.0.255": 234.3,
...

I also verified from the MQTT broker and in Home Assistant that new data is received.

The behaviour is as expected / as it should be from what I can tell. 
Note that the reconnection-timeout is 1 minute between attempts.

Please try again, and let me know if it still doesnt reconnect to the mqtt broker.

Tip: Activate the UDP logging function in the settings menu, then you will receive same debug printouts as the log above to a specified IP/UDP-port. You will need netcat or some other tool. I also have prepared https://github.com/remne/p1ib/tree/master/tools/p1ib_debug.zip which contains an win32 executable that runs a pythonscript that opens a udp port on your computer if you dont have access to some other tool.

BR,
Andreas

NateRobinsonS, 2022-03-08:
According to the log stream from P1ib, it is able to reconnect to the MQTT broker after it starts up again, as expected.  However, HASS doesn't find any information from the P1ib on the MQTT broker until after a restart of the P1ib.  Does it have something to do with autodiscovery?  Is there something that the P1ib publishes when it first connects that it doesn't republish after it loses the MQTT connection?

It's entirely possible that the problem is with HASS and not P1ib.  However, I don't have any problem with the Z-wave JS integration that runs through the same MQTT server.  It reconnects after the MQTT server restarts, exactly as it should.

Can you reproduce the connection problem in HASS?  I'm running 2022.3.1 with the MQTT integration configured via the front end (not via YAML).  I don't find any errors in HASS' logfile, but I only have Warnings and above activated at the moment.

-Nate

remne, 2022-03-08:
You are correct, the p1ib only registeres sensors once during startup. I thought that was good enough, but I can change that behavior of course.

The reasoning for this behavior was that I suppose that normally one doesnt reset all HA configuration that often (or never? I haven't done it for the last 4 years), and that it was good enough to register sensors once at restart of the p1ib.

BR,
Andreas

NateRobinsonS, 2022-03-08:
This one (also) appears to be on my side.  I found a typo in the config file for the MQTT broker.  The persistent data wasn't being saved at shutdown.  After resolving the issue, I can confirm that HA starts importing p1ib data via the broker as soon as the broker is available again (without restarting the p1ib).

closed 2022-03-08 by NateRobinsonS

radhus, 2023-03-14:
Hi @remne , 

> You are correct, the p1ib only registeres sensors once during startup. I thought that was good enough, but I can change that behavior of course.

I think this would be nice, as otherwise we'll have to have persistence on the MQTT server and/or make sure to restart p1ib in case you would reinstall the MQTT server. Maybe not the big of an hassle, but perhaps it could be changed to publish every time you have (re)connected to the MQTT server?

I haven't looked at the code base yet, so not sure how hard it would be. Perhaps I can try to take a look later. Let me know if I should open a new issue instead.

Thanks for this software and hardware, works great!