remne/p1ib #5: Reactive power incorrectly reported

NateRobinsonS, 2022-03-07:
The P1ib is reporting reactive power (power returned to the grid) even though there is no power production or storage available.  These values are reported via both MQTT and the P1ib dashboard (web interface).

Firmware: development build f7b009f.

remne, 2022-03-07:
Reactive power is sent back and forth the power grid, depending on your load (inductive/capacitive and reactive loads, and if they have poor power factors). There are several good pages on the internet that describes what reactive power is.

It is normal to have export of reactive power.
P1IB does not calculate any values by itself, it only parses the information sent from the power meter.

NateRobinsonS, 2022-03-08:
Thanks, Andreas.  I had looked quickly on the internet, but stopped reading at "power returned to the network".

Apparently, there's a very large phase angle, as the reactive power out (-) is about 60% as large as the active power in (+).

Previously, I used a wired connection to the power meter and the "DSMR slimme meter" integration in HASS.  I don't think it reported the reactive power, so I don't know how it varies over time yet.

-Nate