remne/p1ib #6: Wi-fi reconnection

NateRobinsonS, 2022-03-10:
Hej,

What does the p1ib do if the wi-fi connection is interrupted?  From what I can tell, it goes into AP mode and doesn't try to reconnect to the wi-fi (at least within 10-20 minutes) without a restart.  Is this the intended behavior?

-Nate

remne, 2022-03-10:
Hi,

That is correct, if it looses wifi connection and cannot reconnect immediately it will go into AP-mode, and it will be in AP-mode for at least 15minutes. After 15minutes, it will scan SSID's, and if the configured one is found, a new connection attempt is executed.

The reasoning for the (long) 15minutes interval is that otherwise there will be too small window of opportunity to configure the p1ib settings until you are disconnected from AP-mode. I've experimented with less, and there have been some persons that misstyped their password, and then it got a bit difficult to change it.

I'm planning to make all timeout/interval-settings configurable in the web UI.

BR,
Andreas

closed 2022-03-29 by remne

NateRobinsonS, 2022-04-04:
I have the p1ib "installed" about 1 meter from our wifi base station and within close range of a second mesh point.  In other words, wifi connectivity shouldn't be a problem.  However, it loses contact at irregular intervals about once a day, sometimes several times a day.  This wouldn't be an issue if it recovered within a minute or so.  However, losing 15 minutes of data every time there's a hiccup means missing the price change of electricity at the end of the hour quite regularly.  In my opinion, the p1ib needs to attempt to reconnect several times before defaulting to AP mode for 15 minutes as it does now..

If the issue is preventing the user from "locking themselves out" of the p1ib because the wifi settings on the device are incorrect, might I suggest one of the following solutions?
- The P1ib tries to reconnect to the programmed wifi every 10 seconds for 15 minutes, at which point it gives up and returns to AP mode for 15 minutes before trying to reconnect again for 15 mintues, etc.
- The P1ib tries to reconnect continuously, but can be reset to "factory defaults" by disconnecting and reconnecting the power 5 times within one minute (since there's no hardware switch to use for a reset).

-Nate

remne, 2022-04-05:
For sure, things can probably be tweaked a bit. 
But I rather not have it to try much longer than it is today before entering AP mode. That would be frustrating in the case that you actually need it to enter AP mode.

As it is right now, it will retry 3 times after a disconnect, then wait 5 seconds, and then retries 3 times again after each other. This attempt-cycle is done 5 times before entering AP mode. So a total of 15 times over a period of 15 seconds which should be enough time between an AP restart, which I suppose is the most common reason for a wifi disconnect.

Debug log:
WIFI: No wifi connection, reconnecting...
WIFI: Connecting to AP access_point...
WIFI: Retrying connection (attempt 0, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 1, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 2, tx=40, wifi status=1)...
uptime:372s      IP 0.0.0.0      WiFi TX: 34/40
WIFI: No wifi connection, reconnecting...
WIFI: Connecting to AP access_point...
WIFI: Retrying connection (attempt 0, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 1, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 2, tx=40, wifi status=1)...
uptime:377s      IP 0.0.0.0      WiFi TX: 34/40
WIFI: No wifi connection, reconnecting...
WIFI: Connecting to AP access_point...
WIFI: Retrying connection (attempt 0, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 1, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 2, tx=40, wifi status=1)...
uptime:382s      IP 0.0.0.0      WiFi TX: 34/40
WIFI: No wifi connection, reconnecting...
WIFI: Connecting to AP access_point...
WIFI: Retrying connection (attempt 0, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 1, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 2, tx=40, wifi status=1)...
uptime:387s      IP 0.0.0.0      WiFi TX: 34/40
WIFI: No wifi connection, reconnecting...
WIFI: Connecting to AP access_point...
WIFI: Retrying connection (attempt 0, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 1, tx=40, wifi status=1)...
WIFI: Retrying connection (attempt 2, tx=40, wifi status=1)...
WIFI: Retry count exceeded.
WIFI: Switching to AP mode... 
WIFI: Now in AP mode. Connect your station to SSID 'p1ib'.
WIFI: P1IB have IP: 192.168.1.1 

But as I wrote earlier, I will make these parameters configurable in the web ui, so it can be tweaked for custom needs.

However from what you are writing, the root problem seems to be that it actually disconnects at all. This should of course not happen. For example my own doesnt disconnect at all, last time I checked before restarting it had 47 days uptime without any reconnections to wifi.

I dont know if it depends on ESP32's wifi in combination with your AP (or the mesh) is the problem or something else. I would say it have nothing to do with the p1ib software, as it uses the espIDF framework for wifi connectivity.
I do though now that there have been reports with connection issues with certain AP's with the ESP series of wifi mcu's from reading forum posts and suchs.

Might I ask if you have some old AP laying around that you can try with temporary, to rule out your AP's in combination with the ESP32? 
Or disable your AP mesh for a day or two and see it that does the trick?
If that doesnt help, it might be a problem with your p1ib hardware. Bad/glitchy pcb-antenna maybe? I always test all devices before I send them out (wifi connectivitivy, P1 port and cable), but I don't, for obvious reasons, test each device over longer periods of time.

/Andreas

NateRobinsonS, 2022-04-07:
I'm definitely not trying to pin the source of the disconnection problem on
the p1ib (hardware or software).  I'm running Asus' "Zen wifi" XT8 and it
seems to have a few bugs, particularly with "IOT devices".  I suspect that
there is radio interference here with the 2.4 GHz band (some devices
have trouble with Bluetooth connectivity in the house as well).

Instead, I'd like to set the device up to recover quickly.  As you suggest,
a setting to adjust the "timeout" before switching over to AP mode and/or
how frequently it leaves AP mode (presuming that there is nothing connected
via HTTP) to look for the previous wifi would solve this.

Do your APs restart in 15 seconds?  If I restart the Asus mesh, it takes a
few minutes to come back up.

-Nate

NateRobinsonS, 2022-04-07:
> I see, it doesnt sound like the best rf environment!
> Anyhow, if you want to try with another p1ib, your very welcome to fetch
> one when around linköping!
>
> Regarding restart times ,yes, it actually do :). But I agree, its a bit
> optimistic value for most of the products on the market I suppose.
>
> I've added wifi connection retry timeout and the ap mode timeout
> configurable in the web ui in latest dev firmware (sha1 91b79c) and I also
> changed the default retry timeout to 60s.
>
> Let me know if you have any issues with the firmware.
>
> /Andreas

Thanks. I'll try it out.

-Nate

remne, 2022-04-22:
Hi Nate,
Hows it going with your reconnection issues? Did it help out anything with the possibility to modify the timeout values for reconnection etc?

BR,
Andreas

NateRobinsonS, 2022-04-23:
Absolutely.  I set the Wifi Retry Connect timeout to 120 seconds and the
Wifi Wi-Fi Connect Attempt Timeout In AP Mode to 5 minutes.  The device
hasn't been offline for 15 minutes (when a HA automation would send me a
notice) since the update.

Thanks!

-Nate