remne/p1ib #10: How to delete orphan entities?

johanschelin, 2022-06-16:
Hi!

I purchased this, but my Meter didn't support opening the P1 port, so I sold it to a friend.
I have all my entites still in Home Assistant, and cannot find a way to delete them? :)

This entity ('sensor.p1ib_active_power_minus_l1') does not have a unique ID, therefore its settings cannot be managed from the UI. See the [documentation](https://www.home-assistant.io/faq/unique_id) for more detail.

johanschelin, 2022-06-16:
I believe that I should send an empty message with the Retain flag, but I am a MQTT beginner, so I'm not sure what topic i should send it to.
[image]

remne, 2022-06-25:
Hi!
I guess you have one of the earlier firmwares.
For newer ones you just go to "settings -> devices & Services -> Devices" and press "delete device".
[image]

The other way (before the firmware supported the device-entry in HA) is to send an mqtt message to the Home Assistant "config" topic with an empty payload.
I have a script for this, which deletes all sensors. From this you can find the topics. It have changed a bit during the development of the p1ib firmware, so I'm not sure which one you have. Probably the sensornames are the ones in the bottom of the script without the prefix-part.
Probably "homeassistant/sensor/p1ib/p1ib_h_active_imp_q1_q4/config" for example. Try it out and see if it works!

$ cat unregister_all_sensors.sh 
#!/bin/bash

ARG=$1
SUFFIX=""

if [ -z "$ARG" ]
then
    echo "empty suffix"
else
    SUFFIX="${ARG}"
    echo "adding suffix \"${SUFFIX}\" to topics"
fi

set -x

mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}h_active_imp_q1_q4/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}h_active_exp_q2_q3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}h_reactive_imp_q1_q2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}h_reactive_exp_q3_q4/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}active_power_p_q1_q4/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}active_power_m_q2_q3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}reactive_power_p_q1_q2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}reactive_power_m_q3_q4/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}active_power_p_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}active_power_m_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}active_power_p_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}active_power_m_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}active_power_p_l3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}active_power_m_l3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}reactive_power_p_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}reactive_power_m_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}reactive_power_p_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}reactive_power_m_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}reactive_power_p_l3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}reactive_power_m_l3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}voltage_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}voltage_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}voltage_l3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}current_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}current_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib/p1ib_${SUFFIX}current_l3/config" -m ''

mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_h_active_imp_q1_q4/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_h_active_exp_q2_q3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_h_reactive_imp_q1_q2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_h_reactive_exp_q3_q4/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_active_power_p_q1_q4/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_active_power_m_q2_q3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_reactive_power_p_q1_q2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_reactive_power_m_q3_q4/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_active_power_p_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_active_power_m_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_active_power_p_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_active_power_m_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_active_power_p_l3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_active_power_m_l3/config" -m ''
[... cut ...]
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_voltage_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_voltage_l3/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_current_l1/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_current_l2/config" -m ''
mosquitto_pub -h 192.168.0.5 -p 1883 -t "homeassistant/sensor/p1ib_${SUFFIX}/p1ib_current_l3/config" -m ''

johanschelin, 2022-07-09:
Sorry for the late reply

Sending empty payloads to the config topic removes the sensor temporarily. But a restart makes it pop back again. Can I do this in another way? I'm sending the messages via the MQTT Integration.

remne, 2022-07-09:
You need to clear the messages in the mqtt broker probably (they might be stored as retained messages)
Read about it here: 
https://community.openhab.org/t/clearing-mqtt-retained-messages/58221

johanschelin, 2022-07-09:
Excellent, doing this for every topic deleted them correctly - thanks

[image]

closed 2022-07-09 by johanschelin