remne/p1ib #12: Not able to OTA upgrade to fcd8d4a

bckstrm, 2023-04-23:
I have some problems with p1ib not reconnecting after temporary loosing wifi-connection and i wanted to test the development firmware to see if this has been solved but every time i try to upgrade firmware from 97b55c9 to fcd8d4a it fails with error message "Error in server response!".

remne, 2023-04-24:
Hi,
P1IB normally enters AP-mode for a given time (timers can be modified in the web configuration) when it cannot find the configured SSID while loosing the signal. If your wifi-signal is received very poorly, this can happen. It will stay in AP-mode for default 15 minutes, then scan again and reconnect to your ssid (if found).

[image]

But from your question in hand, it sounds like that the upgrade gets interrupted due to poor wifi signal. 
Would it be possible that you move your p1ib close to the wifi-accesspoint temporary (powering it by USB), and retry?

Regards
Andreas

bckstrm, 2023-05-01:
I'm sorry, problem was at my end (VLAN was blocked from internet). I will try if the retry trigger works better with the latest firmware, thanks!

closed 2023-05-01 by bckstrm