remne/p1ib #13: Feature request: composite MQTT messages

albinsunnanbo, 2023-06-11:
Currently p1ib publishes 1 topic per metric.
It would be nice if there would be an option to get a single topic with a composite JSON document with all metrics instead.
This would simplify some calculations and improve performance for some clients.

remne, 2023-06-12:
Hi,
Thanks for the request.
I will add this feature in next coming development firmware.

Regards,
Andreas

albinsunnanbo, 2023-06-15:
Thanks
It would be nice if you could also trim leading zeroes, like in 
[image]
Some libraries are really strict about number formatting

remne, 2023-09-14:
Json payload for sensors states is nov available in dev firmware (2023-09-14, adcc0f7).

closed 2023-10-02 by remne