remne/p1ib #15: Feature request: AWS IoT hub compatibility

albinsunnanbo, 2023-06-15:
I plan to use AWS IoT for analyzing the smart meter data.
I tried and had to write a wrapper layer, but with a few tweaks it works. It would be nice if the p1ib could support those things out of the box.
* Needs JSON payloads, not plain strings - https://github.com/remne/p1ib/issues/13 would solve that issue
* Can't handle leading zeroes, for example in
  [image]
  I suppose this will also be fixed in - https://github.com/remne/p1ib/issues/13
* Needs mutual TLC with X.509 client certificates, a way to upload and use a X.509 client certificate and Root CA to be used - https://docs.aws.amazon.com/whitepapers/latest/device-manufacturing-provisioning/resources-on-the-device.html

Azure IoT hub seems to have similar requirements although it seems to require username and can alternatively use an authentication key instead of X.509 certificate
https://learn.microsoft.com/en-us/azure/iot/iot-mqtt-connect-to-iot-hub#using-the-mqtt-protocol-directly-as-a-device

remne, 2023-07-07:
Hi,
TLS support for MQTT is in progress, will be released in a near future.

/Andreas

remne, 2023-08-04:
Hi! 
Latest development firmware supports MQTT TLS (AWS IoT Core).

However, I have not finished json payloads yet. 
I'm not sure however why you "need" json objects for AWS? Its no problem to run with plain strings to the AWS MQTT broker. But I guess its your specific solution that needs json objects?

BR,
Andreas

LindquistS, 2023-08-23:
The supplier is RemneTech, I am currently looking at their suitability as a supplier.

closed 2023-09-22 by remne