remne/p1ib #16: Technical details regarding Kamstrup OMNIPOWER Three-phase meter DK-8660

christofferholmstedt, 2023-07-06:
I just moved to a newly built house where the electricity company has installed the Kamstrup OMNIPOWER Three-pase meter DK-8660. I see in a lot of web shops for various devices that the OMNIPOWER is not supported, but rarely any details on what exactly is wrong with this specific meter.

I see in the README.md that the P1ib can be modified to support my OMNIPOWER but not recommended. Can anyone here share some more technical details what is actually wrong with the OMNIPOWER and what needs to be modified in the P1ib?

Any help is appreciated before I go down the rabbit hole of trying to figure it out myself.

remne, 2023-07-07:
The omnipower meter dosnt comply to the P1-standard. 
- it doenst have any power supply in the hardware interface
- the baudrate can vary (based on grid owner configuration)
- the payload can differ (based on grid owner configuration)
- sometimes its a mess between how the software is configured against the installed interface-module on the meter, even if the module is installed as a P1-module, the grid owners doesnt know themselfs sometimes or how to configure it correctly to get the expected output. 

...its all a mess.

I've helped a couple of customers of the p1ib to make it work with the kamstrup omnipower, but it takes often several phone calls to the grid owner, both to get hold of a person that knows anything, and some soldering on the p1ib.

The p1ib itselfs needs to be patched with one cable, to power the p1-interface to be powered by usb instead of via the p1 interface. Its not difficult if you have some earlier experience with soldering. The problematic part is to make the grid owner to realize how to configure their meter.

Regards
Andreas

closed 2023-07-12 by christofferholmstedt

orfeasKalipolitis, 2024-03-08:
Hello,

Newbiew with the P1IB here, based in Sweden. 

I missed the part where OMNIA and OMNIPOWER are different unfortunately before I bought the device. But I really like the idea and the implementation so I would like to stick with it.

Could you perhaps let us know what you told the grid owner so that we can take our chances with them directly? In my case, since it is quite a small outfit, I believe I could more easily talk to someone technical and if I can give them a good enough description, end up with a solution.

Thanks in advance. Amazing project!

remne, 2024-03-08:
Hi,
As it requires so much and time effort (both mine and yours, as it always results in me having to put in hours in support mails etc) I would recommend that you return your device instead.
The devices is also sold as not supporting the OMNIA as you've already mentioned.

Sorry for the inconvience.

Regards,
Andreas

orfeasKalipolitis, 2024-03-08:
Hi,

Would it be possible to just give a written description of the info you've had to give them in the past?

On the README it still states that it could be possible to connect to this meter. While I understand it may be complicated, I would like to give it a try.

I also completely understand that you don't want to put too much effort into it, as you know it is really time intensive and difficult to solve for everyone, hence the suggestion of just providing that information as some text that we can take on our own and try to irk out a solution.

Thanks in advance.

Orfeas

ipkryss, 2024-03-11:
I'm wondering the same thing! Also just bought a house and saw that I have an omnipower meeter and that it's not supported. 

I found some other products that does support the omnipower so I wonder how they achieved that considering this one couldn't - I'd much rather support this project compared to the other solutions I've found.

remne, 2024-03-12:
I didnt design the hardware to be non DSMR5/P1 compatible. One can patch the PCB to make it work to power some parts of the hardware to work with the Kamstrup meter. But this is also the reason why I'm hesitant to put out the instructions, because people in general more often than seldom doesnt solder and will in some cases most likely destroy their P1IB.
The other is as mentioned before the problem with grid owners and their limited knowledge of configuring their own meters. I would hope that it has improved since I had to support in this area, since it was some time ago. And most likely people are using their meters more often today with some P1-dongle/device.

But as you both seem very keen on getting this up'n'running, send me an mail at hello@remne.tech, and we can take it from there.

orfeasKalipolitis, 2024-03-13:
Thank you Andreas.
I sent an email :)
P.S. my OmniPower should have the latest firmware installed, I just had the local tech update it.

ipkryss, 2024-03-15:
Thanks for replying Remne, but I will not bother you with this. Seems like too big of a project if soldering is necessary even though I am not afraid of doing some soldering usually.

Is there no chance that it will work out-of-the-box with the latest firmware installed as orfeaskalipolitis mentions?