remne/p1ib #18: Non decimal in current

JelpirT, 2023-08-28:
I got no decimals in current flow? any ide how to get it?
[image]

remne, 2023-09-14:
Hi,
What power meter do you have?
I'm not familiar with that model.

To help out to understand why p1ib decodes the information without decimals I need a sample payload. You can get it by activating the remote debug or hooking up the p1ib to usb and connecting to the serial port with a terminal program (2'000'000 baud rate)

Regards,
Andreas

JelpirT, 2023-09-14:
Hej, jag har denna mätare, vet ej vilken ip debug ska aktiveras till?Skickat från min Galaxy

JelpirT, 2023-09-19:
Edit the answer

remne, 2023-10-02:
hi, please contact hello @ remne . tech instead via email!

closed 2023-10-02 by remne