remne/p1ib #20: Entity state class warnings

johanschelin, 2023-09-21:
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:500
Integration: Sensor (documentation, issues)
First occurred: 18:57:56 (3 occurrences)
Last logged: 19:12:03

Entity sensor.p1ib_telegram_received (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('timestamp') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Entity sensor.p1ib_reactive_energy_export_q3_q4 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Entity sensor.p1ib_reactive_energy_import_q1_q2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

johanschelin, 2023-09-21:
And also this

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:661
Integration: Sensor (documentation, issues)
First occurred: 19:12:03 (10 occurrences)
Last logged: 19:12:03

Entity sensor.p1ib_reactive_power_minus_l1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Entity sensor.p1ib_reactive_power_plus_l2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Entity sensor.p1ib_reactive_power_minus_l2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Entity sensor.p1ib_reactive_power_plus_l3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Entity sensor.p1ib_reactive_power_minus_l3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

remne, 2023-09-22:
For kVar, It must been added recently (I havent updated my home assistant to latest). Thanks for notifying this. Unit 'var' (or reactive power) have not been available previously.
Will update unit in next-coming dev firmware.

BR,
Andreas

ivik, 2024-01-16:
Hi guys

Besides the errors above I have also found these: 

2024-01-16 22:07:17.062 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_energy_import_q1_q2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVArh' which is not a valid unit for the device class ('energy') it is using; expected one of ['MWh', 'GJ', 'kWh', 'Wh', 'MJ']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

2024-01-16 22:07:17.065 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_energy_export_q3_q4 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVArh' which is not a valid unit for the device class ('energy') it is using; expected one of ['MWh', 'GJ', 'kWh', 'Wh', 'MJ']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

remne, 2024-01-16:
Hi,
These log entries are already in the post above, which is the reason for my comment on not fixing them until the correct and proper unit exists in home assistant :).

Please read post :
https://github.com/remne/p1ib/issues/20#issuecomment-1731024672

> I will wait to fix this issue with reactive power, as the unit available/recently added is only "var" (and should be spelled "VAr"), and not kVAr, which is the values that is transmitted from our power meters . kVArh (for reactive energy) is not either added yet, which is very strange.
> 
> However I have fixed the first three issues regarding telegram sensor and reactive energy sensor state classes.
> 
> BR, Andreas

remne, 2024-01-16:
I actually started to do a fork and fixed the issue in HA, but then I didnt have time for a pull request and everything around that. But will have to take care of it at some point as it seems no one else is doing it.

diff:
https://github.com/home-assistant/core/compare/dev...remne:core:dev

remne, 2024-01-16:
...and I just realized I actually put the unit kVArh in the wrong place... oh well, some day... :)

ivik, 2024-01-16:
Sorry, I was looking for this:  ['MWh', 'GJ', 'kWh', 'Wh', 'MJ'] in the above messages. 
And could not see it and thought to add it, but if you already know about it and working on a fix then great. 

Thx for a quick reply!