remne/p1ib #21: Feature request: Make the types of messages that are posted to MQTT configurable

klasrpersson, 2023-09-22:
If i only want to post messages to AWS IoT Core then it is unnecessary to send home assistant specific messages. If i dont accept those messages then it looks like the MQTT connection is lost. Would be nice if i could turn off those and other messages that i dont care about.

remne, 2023-09-22:
The connection is lost due to your policy settings in AWS Core. You can change this behavior. I had the same issue when developing the support for AWS.
However, I see your point that its unecessary to send messages not relevant for the given task.
I will most likely create a disable option for the HA integration / MQTT autodiscovery feature.

BR,
Andreas

remne, 2023-10-11:
Hi.
In firmware 4afb28e there is a ui switch for HA specific messages. If disabled and the json payload switch is enabled - the only mqtt messages sent are the meter json object and the general online/offline state (will) message.

BR,
Andreas

closed 2024-02-27 by remne