remne/p1ib #22: Feature request: Remote restart function

exxoff, 2023-10-10:
I have at times noticed that P1IB sometimes loses the MQTT connection. It's rare but it has happened. A quick reboot solves the issue.

In Home Assistant, I have an automation that notifies me if P1IB hasn't updated in a while. I would like to see a webhook or even a Home Assistant service to restart P1IB in a similar automation.

Cheers!

remne, 2023-10-11:
Hi!
I have got a couple reports about this problem - but I have never been able to reproduce the issue. Until a couple of days ago, when my power provider had a scheduled power outage. 

The problem was in the mqtt reconnect logic, which would, in rare occasions, not work as expected. The P1IB needed to be powered on/not restarted for a longer period of time for the error to present itself (which is seldom the case for me as I try new firmwares on the p1ib on my meter from time to time).

I've corrected it in latest dev-firmware (4afb28e).

Other than that, regarding your suggestions, there is a 'telegram last received'-sensor you can check on if no data has been updated in HA for some time. Also there is a http://p1ib.local/restart webservice one can use.
..But I rather see you dont have to add this extra measures, which wont be the case hopefully with latest firmware :).

Regards
Andreas

closed 2023-10-11 by remne

exxoff, 2023-10-12:
World class, Andreas. I will try the new fw immediately. Thanks!