remne/p1ib #24: Active power L1-3 (solar) show power

Mika-gbg, 2023-11-03:
Hi,

Recently connected the P1ib as I got a new power meter.
I note that Active power L1-3 (solar) show power even though my solar system is not producing due to evening/dark condtions, see picture.
What could be wrong?

Best regards,
Mika
[image]

remne, 2023-11-03:
Hi,
It is nothing wrong. The solar sensors are a combination of export and import. It will be negative and positive values depending on if you are exporting or importing power. Right now it is matching your import, if you compare the values. 
The only thing that might be wrong is the naming with the word 'solar', :).

remne, 2023-11-03:
The "solar" sensors exists for people that dont want to create a template sensor in Home Assistant for the combination of export and import sensors.

Mika-gbg, 2023-11-03:
Great! Thank you for the quick reply!

closed 2023-11-03 by Mika-gbg