remne/p1ib #26: Feature-request: Add telnet (TCP server) for raw telegrams

marcus-aa, 2023-12-03:
For applications that have their own sw decoding of P1 messages the de-facto standard is that the meter interface acts as a telnet TCP server that just echoes the telegrams received from the meter.

It would be very nice if this could be added, as the P1IB then would be compatible also with this way of interfacing to HAN-P1.

It also means compatibility with other software expecting directly connected P1->USB cables 

 Examples;
 
 https://community.home-assistant.io/t/norway-sweden-electric-meter-reading-ams-han/384646
 https://www.home-assistant.io/integrations/dsmr/
 https://github.com/legolasbo/smartmeterBridge
 ser2net
 https://github.com/DavyLandman/p1-esp8266
 https://currently.one/
 https://www.zuidwijk.com/product/p1-reader-ethernet/
 https://www.domoticz.com/
 
 And from https://smartgateways.se/
 
 (in Swedish)

```
Avläsning av smarta mätare via Telnet
Gatewayn har en inbyggd Telnet-server. Denna server erbjuder data från den smarta mätaren oförändrat. Detta gör det möjligt att själv konvertera data från den smarta mätaren till användbara data.
Gatewayn stöder upp till 5 samtidiga Telnet-sessioner. Detta gör det möjligt att använda olika hemautomatiseringspaket som DSMR Reader, Home Assistant och Domoticz samtidigt. Perfekt för användare som vill migrera långsamt till annan hemautomatiseringsprogramvara eller som vill läsa av den smarta mätaren med flera paket samtidigt.
 
```

telnet 192.168.31.193
Trying 192.168.31.193...
Connected to 192.168.31.193.
Escape character is '^]'.
/KAM5

0-0:1.0.0(231203182105W)
1-0:1.8.0(00000850.824*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00000001.882*kVArh)
1-0:4.8.0(00000046.800*kVArh)
1-0:1.7.0(0002.876*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.000*kVAr)
1-0:4.7.0(0000.265*kVAr)
1-0:21.7.0(0000.836*kW)
1-0:41.7.0(0001.713*kW)
1-0:61.7.0(0000.327*kW)
1-0:22.7.0(0000.000*kW)
1-0:42.7.0(0000.000*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.000*kVAr)
1-0:43.7.0(0000.158*kVAr)
1-0:63.7.0(0000.000*kVAr)
1-0:24.7.0(0000.266*kVAr)
1-0:44.7.0(0000.000*kVAr)
1-0:64.7.0(0000.157*kVAr)
1-0:32.7.0(229.9*V)
1-0:52.7.0(230.5*V)
1-0:72.7.0(229.8*V)
1-0:31.7.0(003.9*A)
1-0:51.7.0(007.5*A)
1-0:71.7.0(001.7*A)
!233A

remne, 2024-02-10:
Hi,
Sorry for late reply.
I've added it to my feature request list.

Regards,
Andreas

closed 2024-04-08 by remne

marcus-aa, 2024-04-08:
Super, thanks!

marcus-aa, 2024-11-27:
Does "completed" mean that it is implemented? 

BR /Marcus