remne/p1ib #28: Prepare device before using the HAN-port

robbanl, 2023-12-11:
Hello!

I've just bought the device and would like the opportunity to prepare the device before the meeter is switched. Since it's a remote place, I'm planning to setup the wifi connection and then plug it into the meeter once it's switched. There is a risk that they won't be able to turn the HAN-port on the same day, and I might not be there when it's turned on.

So, basically what I'm wondering, can I power the device using a USB-a power source, do my configurations to the wifi, and then shut it down and just have it laying there offline until the meeter is switched and the HAN-port is activated? Think I can leave a note to the people doing the meeter switch to plug it into the new meter :-)

remne, 2023-12-11:
Hi,
Yes its possible to prepare it in beforehand. Just power it by USB. Then under settings there is a "custom SSID"-field, in which you can fill in your remote accesspoint SSID (as you cannot scan for it in your current location). 

Regards,
Andreas

robbanl, 2023-12-11:
Just what I was looking for ! Thanks for a superfast reply! I'll close this issue.

closed 2023-12-11 by robbanl