remne/p1ib #29: WiFi issue

robbanl, 2023-12-25:
Hello!

I run a Unifi network that consists of 3 WiFi access points. They all broadcast 2 networks, 1 for all my IOT devices (2,4ghz only) and one for the rest. When configuring the P1IB unit to connect to my IOT network it didn't work. Doing a scan showed 3 IOT networks and 3 of my regular network. Was unable to get it to connect.

My solution now was to create another separate 2,4 ghz network that was only broadcasted by the outdoor wifi AP. Then it worked just fine.

Just wanted to inform about this issue and if someone else might have the same problem.

remne, 2023-12-25:
Hi,
That is strange. I use Unifi AP:s (AP AC pro) as well, and have the same setup (multiple AP:s, multiple networks on each). It works without any problems here, and I have multiple P1IB:s as well connected to it ;).

P1IB shows each AP with same SSID multiple times, I've done that by purpose in the firmware, but thinking of changing it as it seems to confuse people sometimes (get questions about it frome time to time). However, the ssid itself is just a string so it really doesnt matter. It has no bearing of what AP it will connect to with same SSID.

Regards,
Andreas / Remnetech

robbanl, 2023-12-25:
Yeah, I checked the source and noticed they all had the same ID, so guessed it wasn't an issue :-) Maybe I'm just to tired right now, but I can check it all again tomorrow :)

remne, 2024-01-06:
Were you able to get it to work with your AP:s?

robbanl, 2024-01-06:
Yepp, it works just fine. But still using the new SSID. I haven't been able any more extensive testing, but I'll give it a go in the next couple of weeks. We can close this issue for now :)

closed 2024-01-06 by robbanl