remne/p1ib #31: Red/Blue Blinking LED

LucasHokerberg, 2024-01-17:
I can't reach my P1IB anymore. The problem occurred a few days after I connected it to an HAN switch (as I have new load balancer for my new EV charger). I suspect it might be due to low power, even though the load balancer is working just fine.

In any way, the P1IB LED is blinking blue a few times, then a brief red blink, then back to blue blinking, and so on. What does this pattern mean?

LucasHokerberg, 2024-01-17:
Update: I connected the unit back directly to the energy meter, but still no joy. It blinks blue, then flashes red, then goes back to blinking blue.

remne, 2024-02-10:
Hi,
With HAN-switch I suppose you mean a P1-switch? HAN is the RJ45 standard, which is not supported by the P1IB.

I'm not sure what has happend. But you should contact the m.nu support in first hand.
Issues in github should be used for firmware related issues in first hand. 
Will close the issue.
Please mail hello@remne.tech if you cannot solve the issue via m.nu.

closed 2024-02-10 by remne

marcus-aa, 2024-02-10:
> Hi, With HAN-switch I suppose you mean a P1-switch? HAN is the RJ45 standard, which is not supported by the P1IB.

No, HAN is a general term for the customer port and covers all the different implementations.

The P1 port is one type of HAN. There are HAN-P1, HAN-NVE, HAN-BSI and a british version with zigbee.

remne, 2024-02-10:
OK, Sure :). 
Often people refer to the HAN-RJ45 standard when mentioning HAN. Just wanted to clarify.