remne/p1ib #34: Invalid state message warnings in HomeAssistant

andersjohansson3, 2024-02-15:
Logger: homeassistant.components.mqtt.sensor
Source: components/mqtt/sensor.py:245 
Integration: MQTT (documentation, issues) 
First occurred: 19:08:38 (23 occurrences) 
Last logged: 19:08:59

Invalid state message `{
    "p1ib_clock_date": null,
    "p1ib_hourly_active_import_q1_q4": 5213.196,
    "p1ib_hourly_active_export_q2_q3": 0,
    "p1ib_hourly_reactive_import_q1_q2": 0.955,
    "p1ib_hourly_reactive_export_q3_q4": 418.856,
    "p1ib_active_power_plus_q1_q4": 4.164,
    "p1ib_active_power_minus_q2_q3": 0,
    "p1ib_reactive_power_plus_q1_q2": 0,
    "p1ib_reactive_power_minus_q3_q4": 0.303,
    "p1ib_active_power_plus_l1": 1.3,
    "p1ib_active_power_minus_l1": 0,
    "p1ib_active_power_plus_l2": 1.342,
    "p1ib_active_power_minus_l2": 0,
    "p1ib_active_power_plus_l3": 1.522,
    "p1ib_active_power_minus_l3": 0,
    "p1ib_reactive_power_plus_l1": 0,
    "p1ib_reactive_power_minus_l1": 0.224,
    "p1ib_reactive_power_plus_l2": 0.052,
    "p1ib_reactive_power_minus_l2": 0,
    "p1ib_reactive_power_plus_l3": 0,
    "p1ib_reactive_power_minus_l3": 0.131,
    "p1ib_voltage_l1": 231.2,
    "p1ib_voltage_l2": 232,
    "p1ib_voltage_l3": 233.5,
    "p1ib_current_l1": 5.7,
    "p1ib_current_l2": 5.7,
    "p1ib_current_l3": 6.6,
    "p1ib_firmware": "e9309dd",
    "p1ib_update_available": "",
    "p1ib_import_export_l1": null,
    "p1ib_import_export_l2": null,
    "p1ib_import_export_l3": "1.522",
    "p1ib_import_export": null,
    "p1ib_rssi": "-78",
    "p1ib_meter": "KAM5",
    "p1ib_mac": "********",
    "p1ib_telegrams_crc_ok": 173538,
    "p1ib_ip": "**.**.**.**"
}`
from 'power/p1ib/sensor_state'

andersjohansson3, 2024-03-02:
So, this continues to be an issue, and it's really filling up my warning logs:

Logger: homeassistant.components.mqtt.sensor
Source: components/mqtt/sensor.py:245
Integration: MQTT
First occurred: 1 mars 2024 kl. 23:04:11 (**36028 occurrences**)
Last logged: 09:10:42

Invalid state message `'{"p1ib_clock_date":null,"p1ib_hourly_active_import_q1_q4":5834.307,"p1ib_hourly_active_export_q2_q3":0,"p1ib_hourly_reactive_import_q1_q2":1.034,"p1ib_hourly_reactive_export_q3_q4":494.37,"p1ib_active_power_plus_q1_q4":0.869,"p1ib_active_power_minus_q2_q3":0,"p1ib_reactive_power_plus_q1_q2":0,"p1ib_reactive_power_minus_q3_q4":0.181,"p1ib_active_power_plus_l1":0.288,"p1ib_active_power_minus_l1":0,"p1ib_active_power_plus_l2":0.387,"p1ib_active_power_minus_l2":0,"p1ib_active_power_plus_l3":0.194,"p1ib_active_power_minus_l3":0,"p1ib_reactive_power_plus_l1":0,"p1ib_reactive_power_minus_l1":0.171,"p1ib_reactive_power_plus_l2":0.053,"p1ib_reactive_power_minus_l2":0,"p1ib_reactive_power_plus_l3":0,"p1ib_reactive_power_minus_l3":0.063,"p1ib_voltage_l1":230.2,"p1ib_voltage_l2":230.3,"p1ib_voltage_l3":231.6,"p1ib_current_l1":1.5,"p1ib_current_l2":1.7,"p1ib_current_l3":1.3,"p1ib_firmware":"aae1e85","p1ib_update_available":"","p1ib_import_export_l1":null,"p1ib_import_export_l2":"0.387","p1ib_import_export_l3":null,"p1ib_import_export":null,"p1ib_rssi":null,"p1ib_meter":"KAM5","p1ib_mac":"********","p1ib_ip":"**.**.**.**","p1ib_telegrams_crc_ok":342309}' `
from 'power/p1ib/sensor_state'

There are a bunch of null values in the warning logs, could it be those that are causing the issue?

remne, 2024-03-08:
Hi,
Yes, it should not be null values.
What firmware are you running? and what meter?

I will take a look at this issue during the weekend, sorry for the late reply.

For now you could try to disable JSON payload in the p1ib settings, in that way it will only send mqtt messages on sensors that have new data.

Regards,
Andreas

andersjohansson3, 2024-03-08:
I have an Electric Meter "KAM5", Meter Mode "Mode-D" and running latest firmware, aae1e85.
I'll try with setting MQTT Json state to false.

remne, 2024-03-08:
Yes KAM5, but what physical meter is it? A Kamstrup OMNIA?

andersjohansson3, 2024-03-08:
Just realized i needed to add that 😊 
Yes, it's Göteborg Energis Kamstrup Omnia e-meter