remne/p1ib #35: Updated HA and P1ib - not working

4integration, 2024-02-27:
Hi,

Updated HA Core to 2024.2.4 and restarted p1ib, no sensors gets `unavailable`.
Updated p1ib to latest stable, I see 30+ MQTT messages being sent to topic like `homeassistant/sensor/p1ib/p1ib_h_active_imp_q1_q4/config`

All sensors are still `unavailable`

Also noted in HA logs

```
2024-02-27 19:24:46.409 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_power_minus_l3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.410 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_power_plus_l3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.417 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_power_minus_l2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.418 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_power_plus_l2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.419 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_power_minus_l1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.420 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_power_plus_l1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.424 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_power_minus_q3_q4 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.426 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_power_plus_q1_q2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVAr' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.433 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_energy_export_q3_q4 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVArh' which is not a valid unit for the device class ('energy') it is using; expected one of ['Wh', 'kWh', 'GJ', 'MJ', 'MWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-02-27 19:24:46.435 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.p1ib_reactive_energy_import_q1_q2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kVArh' which is not a valid unit for the device class ('energy') it is using; expected one of ['Wh', 'kWh', 'GJ', 'MJ', 'MWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

```

remne, 2024-02-27:
Probably case is HA changed how entities are named, depending on your previous version of HA.
https://github.com/home-assistant/core/pull/95159  
In HA, go to 'settings -> device & services" -> devices, scroll to p1ib device and take a look.
If you have duplicated sensors, remove the device (... icon and remove device) then restart p1ib for new registration to take place.

Regarding reactive energy/power warnings. This warning is already reported in https://github.com/remne/p1ib/issues/20, and is not fixable due to HA does not support units that define reactive energy and correct for reactive power.

4integration, 2024-02-28:
Actually tried to remove p1ib device yesterday but no success.
Did it again and restarted p1ib but for some reason I cannot access http://p1ib.local no more :(

I saw it registers in my router's device list (Ubiquiti DreamMachine) and accessing IP doesn't work either.
But then it disappeared again and not showing up what it seems....hmm

4integration, 2024-02-29:
Got it working again

closed 2024-02-29 by 4integration