remne/p1ib #37: Home Assistant loose P1ib EVERY time HA is restarted (Unavailable entities)

4integration, 2024-03-16:
Hi,

**Every time** I restart Home Assistant it looses P1ib entities and they became "Unavailable". 
Restarting P1ib doesn't help but I can see the entities flick for <1sec. 

Removing P1ib device and entities, restarting HA and P1ib several times can make it work again.
Very annoying!

"P1IB Telegram received" is received every ~10 sec

I can see MQTT messages being sent and device/entities gets created but Unavailable

HA Core: 2024.3.1
HA: Supervisor: 2024.03.0
HA Operating System: 12.1
HA Frontend: 20240307.0

P1ib Build: aae1e85
P1ib Build Date: 20240212

4integration, 2024-03-16:
Even after deleting the device I removed all `p1ib` from `core.entity_registry` and `core.device_registry` and rebooted before rebooted P1ib...still no success

remne, 2024-03-16:
Hi,
I can see why this is annoying.
However, I'm not so sure this is a problem in p1ib, but something rather in your environment.

But to be sure, lets troubleshoot a bit and see if we can figure it out.

Some background: P1IB send a registration message at each connection to the mqtt broker for each sensor, which contains information about the sensor and its state topic. It also enables availability on each sensor. The availiability is based on the 'will'-message registered during connection to the broker.
This state sends 'online' during connection, and sends the will message 'offline' (automatically sent by the broker if connection is lost or timed out). 
If HA receives 'offline' it will set all sensors connected to it to unavailable.
Note, this is the case for all sensors - except the timeout sensor.

So that is how the HA/MQTT messaging works for sensor availability.

If you start mosquitto_sub with parameter -t '#' and -v ( to subscribe to all topics), you can verify this. 
After you have started mosquitto_sub, then restart your p1ib. Then you should see something like this:
"
p1ib_a0a3b3409e18/state offline
p1ib_a0a3b3409e18/state online
homeassistant/sensor/p1ib/<sensor>/config {...}
homeassistant/sensor/p1ib/<sensor>/config {...}  ....and so forth for each sensor.
"

And then a few seconds later (depending on your json state setting i p1ib ) you will receive values for each sensor or a json message with all sensor values - if your p1ib is connected to your power meter and receives telegrams.
These messages should occur with your power meters telegram interval (most likely each 10s)

If this is the case, then everything works as expected from p1ib and the issue is in your environment.

You can manually try the availability feature by simulating that the p1ib is online or offline by excuting:
$ mosquitto_pub -t "p1ib_a0a3b3409e18/state" -m "offline" -h <broker-ip>
$ mosquitto_pub -t "p1ib_a0a3b3409e18/state" -m "online" -h <broker-ip>
Replace the p1ib mac with your mac.

If your p1ib doesnt send MQTT messages as described above, then please post your MQTT messaging from mosquitto_sub here and also your p1ib settings so that we can try to troubleshoot further.

Regards,
Andreas

4integration, 2024-03-17:
Hi Andreas,

P1ib do send MQTT messages (both on `homeassistant/#` and `p1ib/#`), I however use RabbitMQ and sniff topics with MQTT.fx.

Now I disabled MQTT in P1ib but can still see some messages @ `homeassistant/status` with `online` or `offline` even though disabled but reflects the status of HA. 
Removed entities as described above in HA, and rebooted HA.
P1ib reports `online`, enable MQTT.
Then ~35 messages appears in `homeassistant/#` and then it starts publishing on `p1ib/#`

So agree that P1ib publish messages and by the latest approach in this post, it started to work again.
But feels like the combination of HA and P1ib isn't flawless
Maybe HA gets confused when P1ib sends messages while HA restarts.

remne, 2024-03-17:
Hi,
"homeassistant/status" is not published by P1IB. It's HA that publishes messsages on that topic. So it doesnt have anything to do with P1IB.

If there was an issue with how P1IB sends messages to HA, there would be a lot more people than you that have the same issue. So I'm still leaning against that the issue is still in your environment at this point.

Do you have retained messages disabled or will message support in your mqtt broker?  Both needs to be enabled.
To verify retained messages enabled: do you get latest published data by the p1ib topics when you connect and subscribe with topic # (all topics) with your sniffer?
You should see a 'p1ib_mac/state online' if P1IB is running. Also you would see the different value of each sensor.
Normally this is enabled by default in mosquitto's broker, which is of course is needed. I dont know how RabbitMQ default config looks like.
Otherwhise if HA subscribes to topics later than when any of the MQTT devices that communicate with HA sends its auto discovery information and its availiability state (like P1IB do) HA will miss it and availability will most likely not change from its initial "unavailable"-state.

To verify 'will' message enabled: In your mqtt sniffer, make sure that, when P1IB is going from MQTT state connected to not connected (for instance, you plug P1IB out). A message should be published to p1ib_mac/state with message 'offline' after a few seconds (or whatever timeout value your broker have configured).

If you can send me your MQTT broker configuration, P1IB settings and MQTT messages from when you restart your HA instance while P1IB is running, then maybe I can figure out the issue you have.
Other than that, I'm not sure what I can do to help you out more.

Regards,
Andreas

4integration, 2024-03-17:
Thanks @remne , looked a bit deeper and guess it is time for me to switch to Mosquitto
I do not think messages are retained, should be configurable. 
Since I have had thoughts of switching to Mosquitto....

Saw from [HA](https://www.home-assistant.io/integrations/mqtt/) 

> Neither ActiveMQ MQTT broker nor the RabbitMQ MQTT Plugin are supported, use a known working broker like Mosquitto instead. There are [at least two](https://issues.apache.org/jira/browse/AMQ-6360) [issues](https://issues.apache.org/jira/browse/AMQ-6575) with the ActiveMQ MQTT broker which break MQTT message retention.

So I start in with that.

closed 2024-03-17 by 4integration

4integration, 2024-03-17:
@remne thanks for your help

Now switched to Mosquitto (pretty easy) and it seems to be working fine :) (at least after one HA reboot)

My config:

```
joacim@server:~$ more /data/disk1/mosquitto/config/mosquitto.conf
allow_anonymous false
listener 1883
listener 9001
protocol websockets

password_file /mosquitto/config/pwfile

persistence true
persistence_file mosquitto.db
persistence_location /mosquitto/data/
```

Bonus is to lower the footprint of MQTT broker even though RabbitMQ wasn't that demanding :)

remne, 2024-03-17:
Glad to hear it worked out!

Good to know that RabbitMQ (currently) have issues with retained messages. 

Regards,
Andreas