remne/p1ib #38: Rate limit data sent to MQTT?

seaside1, 2024-03-27:
Is it possible to put some limit on the update frequency on the messages being sent to mqtt?
For me it should be enough to get the state posted every 10th second or so.

remne, 2024-03-27:
Hi,
No there are no setting for exactly that. However, you can enable reduced mode and set a threshold on your sensors to decrease the amount of data that is sent over MQTT.
You can also disabled certain measurements if your not interested in them to reduce data throughput even more.

Regards,
Andreas

seaside1, 2024-04-03:
Would it be possible to add a setting with a  certain update interval time?
Maybe more peoples are interested in this setting.

remne, 2024-04-08:
I will add it to the future feature list.

Regards,
Andreas

closed 2024-04-08 by remne

bjorne, 2024-10-06:
I second this request. The Home Assistant recorder database becomes very big with the high frequency of reporting. 10s, 30s or 60s would be more appropriate for my use-case.

I only rarely require more frequent updated values and it would be great if p1ib's own UI would still be quickly refreshed even if MQTT when was updated with lower frequency.

remne, 2024-10-09:
You have the option to enable "reduced mode", then you can finetune the bandwidth based on what you are interested in.
[image]

bjorne, 2024-10-09:
The hysteresis threshold will delay the update until a large enough difference has been observed, correct? It might work decently for the measures which strictly increase, e.g. kWh, but ideally I would like the updates to be recorded regularly regardless if the change is small, for example during low usage periods. For momentary values such as kW or V it seems less easy to find a reasonable threshold to give the right update frequency. 

Being able to control the frequency directly just seems like a better lever for my use-case, since it directly translates into number of recordings (and thus data size) across all measures equally.

bjorne, 2024-11-11:
@remne I have now given reduced data mode a shot and have some feedback. Overall it doesn’t seem to match what I want well: all (or a subset of) values af a slower frequency.

First, some UI feedback: I need to press save & restart many times before anything happens. (This is on Firefox iOS - haven’t debugged on a computer yet)

Second, several key measures seem to be missing from the list in reduced data mode. For example, `p1ib_h_active_imp_q1_q4`. So when I enable this mode these values simply stop updating. Why are they not included?

If this mode should work well for me it would probably need to work something more like this:

* I enable reduced data mode
* I enable hysteresis on some measure. For example, something like `p1ib_h_active_imp_q1_q4` with 0.01 kWh. (But maybe exposing the telegram number and hysteresis of 10 would work even better)
* When this threshold is met, a complete state update with all (selected?) measures is propagated to MQTT

But maybe a simple rate limiting would be simpler…

Let me know if I’m missing something.