remne/p1ib #39: What rj12 cable to use

seaside1, 2024-04-08:
Hi! 

I bought a 3 m standard rj12 cable for the p1b. But it does not seem to work.

https://www.kjell.com/se/produkter/mobilt/fast-telefoni/installation/telekablar/modulkablar/telefonkabel-rj12-till-rj12-3-m-p38406

Is there any specific layout of the cable?

marcus-aa, 2024-04-08:
Telephone cables come in 2 variants, straight or crossed. For phones it does not normally matter, but as the P1 port is assymetric it has to be the same that the designer of the interface intended. 

Unfortunately, different P1-readers have chosen differently so some readers need one type and some the other. 

You can probably see that the original cable is different from the new one by holding the plugs side by side and comparing.

I have a complete toolkit for crimping the connectors myself, but that is no help to you... Didn't expect to have any more use from analog telephone tools, but good to have now.

seaside1, 2024-04-08:
Can you point me to a cable that works? Looking at the cable in seems like it is crossed. Most han-meters seem to use straight connection (like the one I have).
Would it work to use a rj12 to rj12 adapter and attach the short cable (crossed) to a longer straight (I guess result would be crossed).

labanskoller, 2024-04-24:
I also realized this the hard way. I got a correct (but quite short) cable when I bought my P1B via m.nu. Didn't you get one, @seaside1?
If you have a crossover cable you can extend it with a longer straight cable with such an adapter as you describe. I have successfully patched my smart meter around 10 meters using normal RJ45 sockets used for Ethernet, and then attached the P1B the last ½ meter with the crossover cable.

I think it should be clearly documented that P1B requires a crossover cable, as many other devices (SlimmeLezer for instance) use straight cables and that's what's easy to find e.g. at Kjell&Co.

seaside1, 2024-05-21:
Missed to answer this. Yes I bought a straight cable and used an rj12 to rj12 adapter.
Maybe would be good to document this with the retailers so you don't end up with a new straight cable that is not working

remne, 2024-05-22:
Hi,
@seaside1  good to hear that your issue was solved.
As already mentioned the cable for P1IB is a cross over cable. 

Normally a longer cable than 0.5m is not needed, as the P1IB device is supposed to be placed a long side the power meter, inside the meter cabinet. If one needs better Wi-Fi reception the P1IB-ext product is meant to fill that gap.

I do however also provide 1.5m cables when ordering from remne.tech if one chooses it. M.nu also provides this free of extra charge (at least at the time of writing) if one asks for it. 

There are a couple of things that one should think of when using their own cable. One is that me or m.nu cannot guarentee the function of the p1ib device, or what it could do with the device over time (think in the line of increased risk of esd, voltage drop, bit error during transmission, reflections/ringing in transmission lines, damage to the p1-interface of the p1ib and/or the power meter for instance depending on placement. 
Then there is also the question of product warranty if one uses their own (and possible own manufactured) cable.

Also I dont know how many hours I've spent in support mails, just to realize that the customer manufactured their own cable with bad crimping or didnt realize that the cable is cross over connected. :)

So from my stand point, I'd rather not have people experiment too much with their own cables. But on the other hand I know people will do it anyways, so its better to support than to shut them off :). I do modification on products as well.
I will update the product description on remne.tech regarding the cable is cross over.

Please close the github ticket if you feel that your issue have been solved.

Cheers,
Andreas / Remnetech

closed 2024-09-17 by remne