remne/p1ib #41: Force update of mqtt values in Home Assistant.

DavidLennartsson, 2024-07-02:
I have other sensors in home assistant that are calculated based on the current sensors in p1ib.
Sometimes the amperage draw is almost static for some time, that means that the calculations won't be updated until the value actually changes.

Could you add a "force_update" option so that the values are updated in Home Assistant even if they values haven't changed?
https://www.home-assistant.io/integrations/sensor.mqtt/

remne, 2024-09-17:
Hi David,
I'll add it to the feature list!

remne, 2024-09-20:
force_update added in latest dev firmware

example:
homeassistant/sensor/p1ib/p1ib_h_active_imp_q1_q4/config {"unique_id":"p1ib_hourly_active_import_q1_q4","name":"Active energy import Q1 Q4","stat_t":"p1ib/sensor_state","device_class":"energy","unit_of_measurement":"kWh","state_class":"total","value_template":"{{ value_json['p1ib_hourly_active_import_q1_q4'] }}","device":{"identifiers":["p1ib_b0b21ca00a68"],"manufacturer":"Remne Technologies AB","model":"P1 Interface Bridge","sw_version":"fdce72e","name":"P1IB","configuration_url":"http://192.168.0.232","connections":[["mac","b0:b2:1c:a0:0a:68"]]},"availability":[{"topic":"p1ib_b0b21ca00a68/state"}],"availability_mode":"latest","force_update":true}

[image]

Please verify and give feedback if possible.

Regards,
Andreas

closed 2024-11-23 by remne