remne/p1ib #42: Device not booting up on active HAN port

mijak99, 2024-07-04:
Hi, 

I have just received my device (https://www.m.nu/sensorer-wifi/elmatare-for-p1-han-port-med-wifi) 

Connected it to a Kaifa MA304H4D with activated HAN-port, but nothing happens. The LED does not light up. 
For clarity: 
-  I am using the original crossed RJ12 cable that was supplied from m.nu
-  The HAN port indicator on the meter display is active, indicating the HAN port is active. Have double-checked with the supplier too.

Since the MA304 is listed as a supported meter, I am assuming something is off? 
Any ideas on how I can troubleshoot this issue? 

If this is the wrong place to ask, or if more information is needed, let me know.

remne, 2024-07-10:
Hi,

MA304H4D is a M-Bus meter, and not P1-meter as far as I know. P1IB only works with MA304 meters that have a P1 interface.
If you take a look at the meter, there is probably a "M-Bus"-label somewhere on it, indicating its not a P1 meter, instead it uses HAN-NVE (Norwegian standard).

Regards,
Andreas

mijak99, 2024-07-11:
Thanks for the response.   FYI, I can confirm that there is an M-bus marking on the meter.

I thought that the Norwegian standard was using RJ45, but this one has an RJ12. 

Anyways, thanks for getting back to me and resolving the issue.

closed 2024-09-08 by remne