remne/p1ib #47: MQTT: topic prefix

merbanan, 2024-09-07:
I have 3 meters and readers sending data to one mqtt server. It would be nice to get the prefix in the message. One could then filter directly on the message content instead of using a lookup on the mac-address.

remne, 2024-09-07:
There is an option to have both the MAC and/or an custom prefix on the topic. It should be sufficient to be able to differentiate between the meters.

Regards,
Andreas

merbanan, 2024-09-07:
Yes. It works fine but it would be nice to use the prefix as id directly instead of do a lookup of the topic or mac. I send everything to influx and it would be nice if the json is as complete as possible. If you only have 1 meter it doesnt matter but I have 3 of them.

remne, 2024-09-13:
What do you mean with id? You mean the prefix should be part of the json object?

merbanan, 2024-09-13:
Right now the json objects are passed around in the data pipeline I use, thus the topic is not passed along with it. So I have to do a lookup on the mac address and then map based on that.

Ideally one should be able to configure p1ib to set a key to a custom value. This would remove the need to do the lookup, which would be nice.

remne, 2024-09-20:
p1ibeth/p1ib/sensor_state:
{   'p1ib_active_power_minus_l1': 0,
    'p1ib_active_power_minus_l2': 0,
    'p1ib_active_power_minus_l3': 0,
    'p1ib_active_power_minus_q2_q3': 0,
    'p1ib_active_power_plus_l1': 1.028,
    'p1ib_active_power_plus_l2': 0.494,
    'p1ib_active_power_plus_l3': 0.047,
    'p1ib_active_power_plus_q1_q4': 1.557,
    'p1ib_current_l1': 4.4,
    'p1ib_current_l2': 2.9,
    'p1ib_current_l3': 0.5,
    'p1ib_custom_id': 'some_custom_id',
    'p1ib_eth_mac': 'b0b21ca25457',
    'p1ib_firmware': 'fdce72e',
    'p1ib_hourly_active_export_q2_q3': 12370.641,
    'p1ib_hourly_active_import_q1_q4': 74714.898,
    'p1ib_hourly_reactive_export_q3_q4': 10871.154,
    'p1ib_hourly_reactive_import_q1_q2': 7610.008,
    'p1ib_import_export': 1.557,
    'p1ib_import_export_l1': 1.028,
    'p1ib_import_export_l2': 0.494,
    'p1ib_import_export_l3': 0.047,
    'p1ib_ip': '192.168.0.218',
    'p1ib_meter': 'Aidon',
    'p1ib_reactive_power_minus_l1': 0.195,
    'p1ib_reactive_power_minus_l2': 0.49,
    'p1ib_reactive_power_minus_l3': 0.12,
    'p1ib_reactive_power_minus_q3_q4': 0.809,
    'p1ib_reactive_power_plus_l1': 0,
    'p1ib_reactive_power_plus_l2': 0,
    'p1ib_reactive_power_plus_l3': 0,
    'p1ib_reactive_power_plus_q1_q2': 0,
    'p1ib_rssi': '0',
    'p1ib_telegrams_crc_ok': 5,
    'p1ib_voltage_l1': 235.7,
    'p1ib_voltage_l2': 236.4,
    'p1ib_voltage_l3': 236.8,
    'p1ib_wifi_mac': 'b0b21ca25454'}
}

[image]

Added a custom id field, which results in key/value id: <string> added to the json payload.

Available in latest dev firmware.

Please verify and give feedback if possible.

Regard,
Andreas

remne, 2024-11-23:
Closing the issue as no feedback was given.

closed 2024-11-23 by remne