remne/p1ib #49: MQTT: kWh readings missing

merbanan, 2024-09-08:
The readings from p1ib/sensor_state is missing meter kWh if I can parse things correctly.

    "1-0:1.8.0": [
      755313,
      755313.12,
      755313.12,
      755313.24,
      755313.36,
      755313.48,
      755313.48,
      755313.6,
      755313.72,
      755313.72
    ],
    "1-0:2.8.0": [
      25996.92,
      25996.92,
      25996.92,
      25996.92,
      25996.92,
      25996.92,
      25996.92,
      25996.92,
      25996.92,
      25996.92
    ],

{
  "p1ib_clock_date": null,
  "p1ib_hourly_active_import_q1_q4": 6294.273,
  "p1ib_hourly_active_export_q2_q3": 216.641,
  "p1ib_hourly_reactive_import_q1_q2": 0.63,
  "p1ib_hourly_reactive_export_q3_q4": 1234.576,
  "p1ib_active_power_plus_q1_q4": 0,
  "p1ib_active_power_minus_q2_q3": 0.012,
  "p1ib_reactive_power_plus_q1_q2": 0,
  "p1ib_reactive_power_minus_q3_q4": 0.139,
  "p1ib_active_power_plus_l1": 0.005,
  "p1ib_active_power_plus_l2": 0.025,
  "p1ib_active_power_plus_l3": 0,
  "p1ib_active_power_minus_l1": 0,
  "p1ib_active_power_minus_l2": 0,
  "p1ib_active_power_minus_l3": 0.043,
  "p1ib_reactive_power_plus_l1": 0,
  "p1ib_reactive_power_plus_l2": 0,
  "p1ib_reactive_power_plus_l3": 0,
  "p1ib_reactive_power_minus_l1": 0.045,
  "p1ib_reactive_power_minus_l2": 0.055,
  "p1ib_reactive_power_minus_l3": 0.039,
  "p1ib_voltage_l1": 230.3,
  "p1ib_voltage_l2": 229.8,
  "p1ib_voltage_l3": 231.5,
  "p1ib_current_l1": 0.2,
  "p1ib_current_l2": 0.3,
  "p1ib_current_l3": 0.2,
  "p1ib_import_export_l1": 0.005,
  "p1ib_import_export_l2": 0.025,
  "p1ib_import_export_l3": -0.043,
  "p1ib_import_export": -0.012,
  "p1ib_rssi": "0",
  "p1ib_telegrams_crc_ok": 169,
  "p1ib_meter": "ELL5\\253876812_B",
  "p1ib_mac": "b0b21ca25630",
  "p1ib_firmware": "f934567",
  "p1ib_update_available": "",
  "p1ib_ip": "0.0.0.0"
}

remne, 2024-09-08:
"p1ib_hourly_active_import_q1_q4": 6294.273,
"p1ib_hourly_active_export_q2_q3": 216.641,
is the energy measurements import and export (obis code 1.8.0 and 2.8.0).
However I think you might mix the /meterData output and the MQTT messages between your P1IB devices? :)

merbanan, 2024-09-08:
[image]

vs

```
{
  "p1ib_clock_date": null,
  "p1ib_hourly_active_import_q1_q4": 6295.953,
  "p1ib_hourly_active_export_q2_q3": 216.642,
  "p1ib_hourly_reactive_import_q1_q2": 0.63,
  "p1ib_hourly_reactive_export_q3_q4": 1235.146,
  "p1ib_active_power_plus_q1_q4": 0.587,
  "p1ib_active_power_minus_q2_q3": 0,
  "p1ib_reactive_power_plus_q1_q2": 0,
  "p1ib_reactive_power_minus_q3_q4": 0.147,
  "p1ib_active_power_plus_l1": 0.207,
  "p1ib_active_power_plus_l2": 0.218,
  "p1ib_active_power_plus_l3": 0.162,
  "p1ib_active_power_minus_l1": 0,
  "p1ib_active_power_minus_l2": 0,
  "p1ib_active_power_minus_l3": 0,
  "p1ib_reactive_power_plus_l1": 0,
  "p1ib_reactive_power_plus_l2": 0,
  "p1ib_reactive_power_plus_l3": 0,
  "p1ib_reactive_power_minus_l1": 0.047,
  "p1ib_reactive_power_minus_l2": 0.05,
  "p1ib_reactive_power_minus_l3": 0.05,
  "p1ib_voltage_l1": 232.1,
  "p1ib_voltage_l2": 231.3,
  "p1ib_voltage_l3": 234,
  "p1ib_current_l1": 0.9,
  "p1ib_current_l2": 0.9,
  "p1ib_current_l3": 0.7,
  "p1ib_import_export_l1": 0.207,
  "p1ib_import_export_l2": 0.218,
  "p1ib_import_export_l3": 0.162,
  "p1ib_import_export": 0.587,
  "p1ib_rssi": "0",
  "p1ib_telegrams_crc_ok": 1709,
  "p1ib_meter": "ELL5\\253876812_B",
  "p1ib_mac": "b0b21ca25630",
  "p1ib_firmware": "f934567",
  "p1ib_update_available": "No",
  "p1ib_ip": "0.0.0.0"
}
```
But I think I see what the issue is. If I add the transformer scaling factor I get somewhat the correct reading. As can be seen in the output I am running the latest stable release.

6295.953*120 = 755514.36

But I also see that the mac address miss-matches by 3.

merbanan, 2024-09-09:
With a58dff6 this issue is fixed.

The mac address is still different compared to the ui though.

closed 2024-09-09 by merbanan

remne, 2024-09-10:
The MAC in the json message is actually the Wi-Fi MAC. The ethernet MAC is not added in the json message in the current firmware, which might be a bit confusing.
I will add the ethernet mac as well in the next firmware.