remne/p1ib #53: Some state values no longer update in MQTT

bjorne, 2024-09-17:
Hi,

I've been running the device for a while and love it. However, I noticed that in Home Assistant the Energy dashboard would just show zero energy usage. After digging around it seems like the active energy import state is not being properly updated in MQTT. It does show a continuously updated value in the P1IB interface, though. 

The same goes for a few other fields like the momentary voltage. Other values like the momentary power seem to update just fine.

Here is an example:

[image]

In Home assistant it looks like this:

[image]

As you can see, the HA value is ~402 kWh whereas in P1IB it's showing ~1886 kWh, so it's been stuck for a longer time.

The following screenshot from MQTT Explorer shows the values that update continuously highlighted, and most of the ones not highlighted seem stuck:

[image]

I would like to try to factory reset the device, but let me know if you would like more information to troubleshoot this issue before I do.

remne, 2024-09-17:
Hi,
It looks like you have activated "json state" in your P1IB settings, which probably explains why no messages are sent on the per sensor topics.
[image]

In your image its not visible, but at least the sensor_state topic is there with the json payload.

Regards,
Andreas

bjorne, 2024-09-17:
Thanks for the suggestion. I should have mentioned I tried to enable this briefly to try and see if there was any difference but the problem seems to exist also in this state, i.e. only some of the properties in the state have a value. The screenshots I shared was from when I had disabled it again (but the topic remains in mqtt, I guess.)

bjorne, 2024-09-17:
Not sure that my point was clear above. When I enable the feature, the JSON state looks like this:

```json{
  "p1ib_hourly_active_import_q1_q4": "",
  "p1ib_hourly_active_export_q2_q3": "",
  "p1ib_hourly_reactive_import_q1_q2": "",
  "p1ib_hourly_reactive_export_q3_q4": "",
  "p1ib_active_power_plus_q1_q4": "",
  "p1ib_active_power_minus_q2_q3": "",
  "p1ib_reactive_power_plus_q1_q2": "",
  "p1ib_reactive_power_minus_q3_q4": "",
  "p1ib_active_power_plus_l1": "",
  "p1ib_active_power_minus_l1": "",
  "p1ib_active_power_plus_l2": "",
  "p1ib_active_power_minus_l2": "",
  "p1ib_active_power_plus_l3": "",
  "p1ib_active_power_minus_l3": "",
  "p1ib_reactive_power_plus_l1": "",
  "p1ib_reactive_power_minus_l1": "",
  "p1ib_reactive_power_plus_l2": "",
  "p1ib_reactive_power_minus_l2": "",
  "p1ib_reactive_power_plus_l3": "",
  "p1ib_reactive_power_minus_l3": "",
  "p1ib_voltage_l1": "",
  "p1ib_voltage_l2": "",
  "p1ib_voltage_l3": "",
  "p1ib_current_l1": "",
  "p1ib_current_l2": "",
  "p1ib_current_l3": "",
  "p1ib_firmware": "f934567",
  "p1ib_update_available": "",
  "p1ib_import_export_l1": 0.042,
  "p1ib_import_export_l2": 0.196,
  "p1ib_import_export_l3": 0.148,
  "p1ib_import_export": 0.386,
  "p1ib_rssi": "-90",
  "p1ib_meter": "KAM5",
  "p1ib_mac": "b0b21ca2350c",
  "p1ib_clock_date": "2024-09-17T11:18:41Z",
  "p1ib_telegrams_crc_ok": 40,
  "p1ib_ip": "192.168.66.37"
}
```

I assume that all the values should be populated?

The diff between two messages looks something like this (with 3-5 updated values)

[image]

Not sure if this helps narrow it down, but to me it seems to suggest that the reported state is missing some values regardless of whether the JSON state is being used or not?

bjorne, 2024-09-18:
I went ahead and did a factory reset. Reconfigured the same MQTT settings and it is now updating all values as expected. Feel free to close this unless you want to investigate it further.

remne, 2024-09-20:
Its strange indeed. I'm not sure how this error could occur. I've tried to reproduce the problem, but wasn't successful. 
Will close the ticket for now, lets see if it pops up again.

closed 2024-09-20 by remne