remne/p1ib #54: MQTT state does not update

MagnusPer, 2024-10-16:
Today I got my new P1IB, great product! My use-case is to get current measurements via MQTT but the states does not updates as shown in local dashboard. Looking in number of messages in MQTT explorer shows no updates!

Firmware version: Build 54aa555

[image]

MagnusPer, 2024-10-16:
After some reading in closed issues I found this one https://github.com/remne/p1ib/issues/53 which lead me to do a factory reset. Since then it works perfectly 

[image]

remne, 2024-10-17:
Hi, 
Thanks for reporting this issue. 
I need to dig deeper into this, as you are the second user in a relative short time that have experienced this.

Regards,
Andreas

remne, 2024-11-23:
I think you might have enabled "reduced mode", which in your firmware version sends the clock_date and the calculated power metrics, but not the other ones.

This was the case with the other customer as well.
I will close the ticket, but if you see this issue again in the future, please reopen the issue.

closed 2024-11-23 by remne

bjorne, 2024-11-24:
I just wanted to clarify that reduced data mode was not the issue in #53 - and doesn't seem to apply here either? It looks to me like we hit the same problem where some state stopped updating without explanation and the only known fix is a factory reset.

remne, 2024-11-24:
It is the reduced mode. 
But I still dont know why it was activated (even if the webui didnt show it as enabled or not).