remne/p1ib #56: Active energy import Q1 Q4 missing in reduced data

bitpwr, 2024-11-18:
I had reduced data enabled with a few metrics selected, including the most important `Active energy import Q1 Q4`. After upgrade to version 54aa555, that metric is not visible on that page and cannot be selected. All these settings were wiped at upgrade and I cannot enable that missing one. The metric is not sent at all when reduced data is enabled.

[image]

bjorne, 2024-11-21:
I also noticed this recently, mentioned it here: https://github.com/remne/p1ib/issues/38#issuecomment-2469106122

remne, 2024-11-23:
Hi,
It misstakenly got hidden at to some refactoring (I moved around the internal structures for the sensors representations). 
Will be restored in next firmware release.

Sorry for the inconvenience.

remne, 2024-11-27:
The energy import metrics are now back in the latest dev firmware.
Please try it out and give feedback if possible.

Thanks.

Regards,
Andreas