remne/p1ib #57: Request for new feature: automatic restart of p1ib

kristoffer262, 2024-11-23:
Since you sometimes need to manually restart the p1ib, it would be nice to have a scheduled restart that could be activated in the GUI. E.g., possibility to set an interval once per week during night where the p1ib will be restarted.

remne, 2024-11-23:
Why do you need to restart it?
Normally it doesnt need to be restarted. The longest I've had a p1ib running is over 6 months. And the reason for restarting it was that I changed firmware on it.

Can you describe why you need to restart it?
You can run http://p1ib.local/restart to force it to restart, but it should not be needed.

kristoffer262, 2024-11-28:
Sure! My p1ib ”hangs” roughly once every senond month. I restart it and everything works again.

closed 2024-11-28 by remne