remne/p1ib #59: MQTT rate limit not working

bjorne, 2024-11-27:
Thank you for adding the rate limit feature. I think the way you have implemented it will be very useful for me.

I haven't been able to make it work, though, it keeps posting to MQTT every second.

I have tried three different values: 10, 20, and 30 (with save+restarts in between of course.)

[image]

I have also verified that the config actually contains the value I have set by inspecting the network response for `getConfig` from the browser when loading the Dashboard:

[image]

remne, 2024-11-27:
I had at as a sub-option under reduced mode first, but changed my mind in the last minute and moved it as a generic option for mqtt and I missed to remove a condition that rm had to be enabled.
I've corrected it in the latest dev firmware (5e4898f).

bjorne, 2024-11-27:
With 5e4898f it works as expected, thanks again for the addition!

closed 2024-11-27 by bjorne