remne/p1ib #60: MQTT not always reconnecting after router reboot

akasubi, 2025-01-31:
On some occasions, p1ib will not reconnect to Home Assistant's MQTT broker after a network outage (scheduled router reboot in my case). The web page works, but no MQTT messages are received by Home Assistant. If p1ib is rebooted from the settings page, it will reconnect to HA's broker.

remne, 2025-09-05:
Hi,
It should reconnect automatically. I've tried to simulate this but I could not get it in a state where it didnt reconnect.
Can you provide debug logs via the usb/remote debugging if you're able to get it in this error state again?

I've tried by killing my wifi-AP in combinations with the MQTT broker. Here it reconnects as expected in all cases.

Regards,
Andreas

remne, 2025-09-13:
I'll close the issue for now. 
Please re-open if the problem should arrise again.
Please also include firmware version in the issue.

Regards,
Andreas

closed 2025-09-13 by remne