remne/p1ib #64: The unit of measurement kVArh and kVAr not valid

anhead, 2025-06-18:
Hi I get this warning in my home assistant mqtt log

Logger: homeassistant.components.mqtt.sensor
Source: components/mqtt/sensor.py:141
integration: MQTT (documentation, issues)
First occurred: 10:55:18 PM (10 occurrences)
Last logged: 10:55:18 PM

The unit of measurement `kVArh` is not valid together with device class `energy`. this will stop working in HA Core 2025.7.0
The unit of measurement `kVAr` is not valid together with device class `reactive_power`. this will stop working in HA Core 2025.7.0

remne, 2025-09-05:
Hi, 
HA has recently added reactive energy units to its constant list (const.py) - it was not available since some time ago, thus I just "faked" them in the sensor metadata in p1ib's firmware. 
I haven't noticed that it was added in HA.

I will push a firmware with updated metadata for the reactive energy sensors during the weekend.

Regards,
Andreas

4integration, 2025-09-14:
Seems to work fine 👍👍👍