remne/p1ib #66: Feature request: power tariff helper

janakelarsson, 2025-08-17:
My provider charges a power tariff: I pay a fee proportional to the average of the five highest power-consuming hours in the month. The provider tells me the meter calculates the power consumption per hour and selects the five highest. Aidon 6534.

I will likely write my own helpers within HA to keep track of this but it would help to have the actual numbers used by the provider.

janakelarsson, 2025-09-05:
If the data is not available in the meter it is better to do it in HA. I was going to close the issue but perhaps you could make use of the diagram helpers I have made in HA so far, will post those here when I have time.

remne, 2025-09-05:
OK, so just for the sake of it, another issue reported ( https://github.com/remne/p1ib/issues/67 ) it seems that power meters in sweden is starting to output more data than is specified by the "p1 standard" in sweden. 

I haven't come across any meter that actually add data for it in sweden - until today. 

So to be sure, if you're willing, lets check what your meter outputs.

Can you try to get some debug logs from your P1IB? By following this guide, https://remne.tech/p1ib-guide/ in the chapter about debug logs.

Regards,
Andreas

janakelarsson, 2025-09-06:
Site seems to be down

janakelarsson, 2025-09-06:
Second cards, shows hourly consumption for 31 preceding days. The selection tools in the UI allows you to select one or more to show. Code is long, I couldn't find a simple way to shorten the series entity list.

```
type: custom:config-template-card
entities: sensor.p1ib_active_energy_import_q1_q4
grid_options:
  columns: full
variables:
  date: new Date();
  month:
    - " Jan"
    - " Feb"
    - " Mar"
    - " Apr"
    - " Jun"
    - " Jul"
    - " Aug"
    - " Sep"
    - " Oct"
    - " Nov"
    - " Dec"
card:
  type: custom:apexcharts-card
  experimental:
    disable_config_validation: true
  header:
    show: true
    title: Förbrukning per timme
  graph_span: 24h
  span:
    start: day
  now:
    show: true
  all_series_config:
    entity: sensor.p1ib_active_energy_import_q1_q4
    extend_to: now
    show:
      legend_value: false
      offset_in_name: false
      extremas: max
    statistics:
      type: change
      period: hour
  yaxis:
    - min: 0
  series:
    - offset: "-0d"
      name: >-
        ${ date.setDate(date.getDate()-0); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-1d"
      name: >-
        ${ date.setDate(date.getDate()-1); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-2d"
      name: >-
        ${ date.setDate(date.getDate()-2); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-3d"
      name: >-
        ${ date.setDate(date.getDate()-3); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-4d"
      name: >-
        ${ date.setDate(date.getDate()-4); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-5d"
      name: >-
        ${ date.setDate(date.getDate()-5); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-6d"
      name: >-
        ${ date.setDate(date.getDate()-6); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-7d"
      name: >-
        ${ date.setDate(date.getDate()-7); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-8d"
      name: >-
        ${ date.setDate(date.getDate()-8); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-9d"
      name: >-
        ${ date.setDate(date.getDate()-9); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-10d"
      name: >-
        ${ date.setDate(date.getDate()-10); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-11d"
      name: >-
        ${ date.setDate(date.getDate()-11); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-12d"
      name: >-
        ${ date.setDate(date.getDate()-12); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-13d"
      name: >-
        ${ date.setDate(date.getDate()-13); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-14d"
      name: >-
        ${ date.setDate(date.getDate()-14); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-15d"
      name: >-
        ${ date.setDate(date.getDate()-15); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-16d"
      name: >-
        ${ date.setDate(date.getDate()-16); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-17d"
      name: >-
        ${ date.setDate(date.getDate()-17); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-18d"
      name: >-
        ${ date.setDate(date.getDate()-18); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-19d"
      name: >-
        ${ date.setDate(date.getDate()-19); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-20d"
      name: >-
        ${ date.setDate(date.getDate()-20); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-21d"
      name: >-
        ${ date.setDate(date.getDate()-21); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-22d"
      name: >-
        ${ date.setDate(date.getDate()-22); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-23d"
      name: >-
        ${ date.setDate(date.getDate()-23); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-24d"
      name: >-
        ${ date.setDate(date.getDate()-24); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-25d"
      name: >-
        ${ date.setDate(date.getDate()-25); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-26d"
      name: >-
        ${ date.setDate(date.getDate()-26); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-27d"
      name: >-
        ${ date.setDate(date.getDate()-27);
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-28d"
      name: >-
        ${ date.setDate(date.getDate()-28);
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-29d"
      name: >-
        ${ date.setDate(date.getDate()-29); 
        date.getDate()+month[date.getMonth()-1] }
    - offset: "-30d"
      name: >-
        ${ date.setDate(date.getDate()-30); 
        date.getDate()+month[date.getMonth()-1] }

```

janakelarsson, 2025-11-17:
The debug output from my device follows the spec exactly